sparkfun/SparkFun-Eagle-Libraries #70: Possible conversion to an OSS-based library?

q2dg, 2015-01-05:
Eagle is closed-source and not free. There are some alternatives to it (http://fritzing.org/about/comparison). Could be possible to fork your incredible library to support some of these alternatives?
Thanks.

ToniCorinne, 2015-02-13:
Hey there - this is actually something we think about quite often. Unfortunately, we haven't found an OSS solution that has all the capabilities for pcb design. Currently, Eagle does have a freeware version [here](http://www.cadsoftusa.com/download-eagle/freeware/) which allows you to at least access all of the .brd and .sch files for our projects (which is why we are still using it). We've been trying to push them to go more open source, and keep hoping the community will keep improving the current OS pcb CAD systems available. It's a project we'd love to support, we simply don't have the resources right now to spearhead it ourselves.

dragonet80, 2015-02-14:
KiCad does not have all requirements?
http://www.kicad-pcb.org/display/KICAD/KiCad+EDA+Software+Suite
 El dia 14/02/2015 0:39, "Toni Klopfenstein" <notifications@github.com> va
escriure:

ToniCorinne, 2015-03-19:
A lot of work has gone into it recently, but it still needs some work before it becomes a viable option for us as a company to use (and to put the time and resources into transferring all of our files over). However, there has been a lot of work towards improving KiCAD and other open source CAD programs through the Open-Source Hardware Association mailing list. You can learn a bit more about that [here](https://www.sparkfun.com/news/1779).

closed 2015-03-19 by ToniCorinne