sparkfun/SparkFun-Eagle-Libraries #72: POWER_JACK_SMD package has flipped GND/GNDBREAK

jesionaj, 2015-02-14:
On the POWER_JACK device, the POWER_JACK_SMD package has pins flipped. The GND symbol pin is connected to P$4, and GNDBREAK is connected to GND. This is backwards, both on the names and based on my testing with the actual part. G$1.GND should be connected to GND and G$1.GNDBREAK should be connected to P$4.

z3t0, 2015-03-30:
@jesionaj Do you mean POWER_JACKSMD under Sparkfun-Connectors?

jesionaj, 2015-03-30:
I see it as POWER_JACK_SMD under the POWER_JACK device, but yes, it's Sparkfun-Connectors, sorry about not mentioning that.

z3t0, 2015-03-30:
@jesionaj Okay thanks, I'll see what I can do. Also I found the one you mentioned, "POWER_JACK_SMD"

z3t0, 2015-03-30:
@jesionaj This might be a stupid question, but if both the part on eagle and the physical component is flipped then wouldn't it be counter intuitive to change the part on eagle if they already have stock of the part 'flipped'? I see what you mean but I do not really get any point in changing it. If I misunderstand what you are saying please correct me.

jesionaj, 2015-03-30:
The physical parts themselves are not flipped -- what I meant is my experience with the actual part showed that the GND/GNDBREAK designations were not correct. I had to insert a jumper between GND and GNDBREAK to make it work on my board.