sparkfun/SparkFun-Eagle-Libraries #73: LED-RGB-CA-THRUCLEAR part is bad

mixographer, 2015-03-05:
Hi,

the LED-RGB-CA-THRUCLEAR is a bad part. it connects all the pads together which makes it a dead short to ground.

mixographer, 2015-03-05:
I bet the diffused one is bad too!

EagleWorks, 2015-03-05:
This is not an issue with the part*, but rather with your current Design Rules.  In short, change the minimum values to 0 (zero).  For more information, see: [sparkfun comment on adjusting restring value when dealing LED-RGB-CA/CC](https://www.sparkfun.com/products/10821#comment-50e49ae3ce395fd575000000)
Note that this may negatively affect some parts that rely on a user's DRU to give adequate pad sizes - *those* would be the bad parts.

* That said, I believe were per-part DRU overrides - which would solve this issue going forward regardless of DRU, merely triggering warnings in the DRU check -  were on the Eagle CAD wish list at some point.

Scott216, 2015-03-05:
I've used this part on PCBs and it actually comes out fine.

mhord, 2015-03-05:
As EagleWorks says, this is a DRU issue.

This is an infuriating aspect of EAGLE- not all of a design's parameters are local. It means that we (SparkFun) need to be **super** careful with our DRU files. It's bitten us in the past and will likely bite us in the future. :-(

closed 2015-03-05 by mhord