sparkfun/SparkFun-Eagle-Libraries #76: QRD1114 Pins 3 and 4 reversed

z3t0, 2015-03-30:
Hi I noticed that the pins 3 and 4 are in the wrong places. The numbers should run clockwise.
https://www.fairchildsemi.com/datasheets/QR/QRD1114.pdf

I have tried to edit it myself but have had no luck. 
[image]

EagleWorks, 2015-03-30:
Reviewing this gave me a minor head-ache.  The highlighted parts in the symbol are fine - pin 3 is the Anode and pin 4 is the 'K'athode in both the library symbol, and the datasheet.

Things get confusing from there.

In the datasheet, the symbol is drawn such that the Anode (pin 3) and the Emitter (pin 2) are on the same side.  In the library, the led, or transistor*, is effectively flipped.  That's not a big issue, as long as the symbol's connections match up with the pads on the package.

Looking at the symbol in the datasheet, it is numbered, clockwise starting from the bottom-left: 1, 2, 3, 4.  However, you're looking at a bottom view, so the actual order if you looked at it in a top view would be 1, 4, 3, 2.  This is also the numbering in the package in the library.

But then you check out the connections in the Device, and it suggests that the Emitter pin (G$1.E) is connected to pad 1, and the Collector (G$1.C) pin is connected to pad 2.  According to the datasheet, this appears to be flipped.

EXCEPT..if you open up the symbol, and inspect the pins (properties), then you'll notice that the pin by the Emitter of the transistor is actually named C, while the pin by the Collector of the transistor is actually named E.  So the correct pins are connected to the correct pads - but they're named wrong.

At least.. I think.  I'm hesitant to touch this and get a pull request going - this needs a closer look :)

z3t0, 2015-03-30:
@EagleWorks Agreed. This also gave me a total headache so I thought I'd just report it here. I'll take a look at the datasheet and try and figure this out. It's the 2 and 3 that confused me in the first place. One of the things I noticed was that the numbers (1, 2, 3, 4) did not show up when I tried to edit the Symbol of the library. Any ideas?

EagleWorks, 2015-03-30:
The numbers by the symbol on the Device editor show which pads the pins are connected to if the pin's 'Visible' state is is set to 'pad' or 'both' - so that wouldn't show in the Symbol editor.

The pin's own names would be visible if the pin's 'Visible' state is set to 'pin' or 'both'.

You would then see the swapped C/E in the Device editor as well :)

z3t0, 2015-03-30:
@EagleWorks Thanks! Now on to the mind numbing work...

z3t0, 2015-04-05:
Okay... I can't really figure this out. Any help would be appreciated. As @EagleWorks has pointed out things are flipped? (I think)...  The transistor seems to be on the wrong way on the symbol?

ToniCorinne, 2016-12-13:
Thanks for pointing that out! I have modified the symbol to match the datasheet, and verified the pin connections. Thanks!

closed 2016-12-13 by ToniCorinne