sparkfun/SparkFun-Eagle-Libraries #77: MISO and SS pins are mislabeled on ATmega32u4

durkie, 2015-03-31:
According to `variants/leonardo/pins_arduino.h` as well as physical confirmation with a multimeter, MISO is digital pin 14 and SS is digital pin 17. These are incorrectly labeled as being on 17 and 14 respectively in the eagle library.

I also have an outstanding pull request from almost 3 months ago correcting analog inputs 8 and 9, which were also switched.

dragonet80, 2015-03-31:
Hello Craig, I also use Sparkfun Libraries. I have "ATMEGA32U4TQFP
(ATMEGA32U4)" from "SparkFun-DigitalIC" library and MISO is labeled as D14
and SS is labeled as D17.

durkie, 2015-03-31:
Weird! I just downloaded the SparkFun-DigitalIC library from github literally 2 minutes ago, did "Open->Library..." in Eagle, and get this: 

[image]

So I have no idea what's going on there.

Is A8 labeled as being on D9 in your library as well? That's another error that I thought I was fixing with my pull request...

dragonet80, 2015-03-31:
I have checked it in my project and I was using a prior version (copy
pasted from an other project) wich is correct:
(see attached file)

I have opened the actual library "SparkFun-DigitalIC", and you are right,
those pins are swapped.

BrentWilkins, 2015-04-05:
@durkie looks like you are correct.  I've updated it.

closed 2015-04-05 by BrentWilkins