sparkfun/SparkFun-Eagle-Libraries #99: TRACE_ANTENNA-2.4GHz_25.7MM has a 0.00 width??

bsodmike, 2016-07-26:
Hi all, there's a serious issue with the part `TRACE_ANTENNA-2.4GHz_25.7MM` in the (RF library)[https://github.com/sparkfun/SparkFun-Eagle-Libraries/blob/master/SparkFun-RF.lbr].  This is the trace antenna used for the SFE Thing.  The part package has a trace width of 0.00 (truncated to 2d.p.)  This causes infinite looping when attempting to export Gerbers.

Thoughts? @jimblom appreciate any help please.

jimblom, 2016-07-26:
Hey @bsodmike - Thanks for the heads up. We do have a fix for this on the [2016_cleanup](https://github.com/sparkfun/SparkFun-Eagle-Libraries/tree/2016_cleanup) branch, which should be getting pulled into master within the next few weeks.

Upping the width of the polygon to 0.1mil helps to prevent cam'ing issues, while avoiding any adverse effects on the antenna shape.

closed 2016-07-26 by jimblom

bsodmike, 2016-07-26:
Ahh thanks @jimblom, my change was 0.2mil; I'll make that adjustment asap.