sparkfun/SparkFun-Eagle-Libraries #111: Fiducial packages need tKeepout (aka courtyards) of at least 10 mils clearance

lewispg228, 2016-12-15:
AOI has a much easier time finding fiducials that have a proper "ring" of clearance around them.

This also requires that we start running the DRC with the tKeepout layer actually on.

lewispg228, 2016-12-15:
Here a recent example that will cause the AOI to occasionally miss a fiducial:

[image]

ToniCorinne, 2016-12-15:
@lewispg228 You can update that package!

lewispg228, 2016-12-15:
On second thought. I think this could work better if it was on tRestrict, this way it will throw DRC errors even if you have that layer turned off (I think).

lewispg228, 2016-12-15:
@ToniCorinne I definitely will get to this, just made an issue to remember :)