sparkfun/SparkFun-Eagle-Libraries #114: Resistors/Capacitors won't add to sketch (Autodesk Eagle 8.0, Win 10 x64)

HoarseHorace, 2017-02-09:
It appears that any device that contains a '%' symbol is causing issues when trying to add a device to a sketch through the control panel, which seems to effect resistors and capacitors primarily. Using 'add' from the sketch works just fine, but something is strange when adding from the control panel.

@SparkFun-Resistors.lbr > 10KOHM

When first selected, the preview on right shows device/package preview and add button inline. If another device is selected, the preview goes blank. if 'Add is selected' an error is thrown.

"Device not found: 10KOHM-HORIZ-1/4W-1%25@C:/EAGLE 8.0/lbr/SparkFun-Eagle-Libraries-master/SparkFun-Resistors.lbr"

andyolivares, 2017-09-28:
Same here! Can't add any capacitor with a '%' symbol.

Also, I'm missing the generic CAP & CAP_POL devices that used to be available from SparkFun-Passives.lbr

gozhev, 2018-01-25:
The issue is still here on Linux x64 Eagle 8.5.2.
Replacing all % symbols with some text (e.g. _PC_) in the .lbr file resolves the issue.

mixographer, 2022-01-03:
I have this issue as well.