sparkfun/SparkFun-Eagle-Libraries #122: Pro Micro incorrect pin assignments

japalmer, 2017-03-24:
Schematic for Pro Micro presents the following pin mappings:
**pin  /  label**
14      11(MOSI)
15      12(MISO)
16      13(SCK)

However, silkscreen and Sparkfun Arduino board manager maps as follows:
**pin  /  label**
14       16(MOSI)
15       14(MISO)
16       15(SCK)

[image]
[image]

I've made updates and submitted a pull request.

PaulZC, 2021-07-03:
Hi @japalmer ,
It took us a while, but we have now merged your PR #121.
Thanks again for identifying this,
Paul

closed 2021-07-03 by PaulZC

PaulZC, 2021-07-03:
PS: It now looks like this:

[image]