sparkfun/SparkFun-Eagle-Libraries #127: DB9 connector is wrongly named/pin 1 marked

TabajaraLabs, 2017-08-18:
In DB9 connector, there is a DB9-male but not DB-9 female. Also, in DB-9 male, the square pin (which should be pin 1) is pin 5, leading to confusion. I modded the library for the correct position and created a pair of new DB9 connectors which are cut to be smaller (I use it a lot, and see it a lot on the net). Feel free to add to the main library.

GitRayman, 2019-11-08:
I think I am seeing this too...  the female DB9 is OK, but the male one is backwards...

PaulZC, 2021-07-03:
Hi @TabajaraLabs / @GitRayman ,
It has only taken us four years (!) but this issue has now been fixed - in [this commit](https://github.com/sparkfun/SparkFun-Eagle-Libraries/commit/093cbd32511d1693b28c254ecff02bd031fec810).
Thanks for identifying this problem,
Paul

closed 2021-07-03 by PaulZC

TabajaraLabs, 2021-07-05:
It happens, at least it was corrected =)