sparkfun/SparkFun-Eagle-Libraries #131: LED fails to load in 8.6.0 - incorrect HTML

uzytkownik, 2018-01-25:
```
Loading <...>/SparkFun-Eagle-Libraries-master/SparkFun-LED.lbr ...

Error:

line 2049, column 2: Unexpected '<'.
```

It looks like incorrect git merge:

```
<<<<<<< HEAD
<package name="APA102">
<smd name="2" x="-2.1" y="0" dx="1.2" dy="1" layer="1"/>
<smd name="1" x="-2.1" y="1.6" dx="1.2" dy="1" layer="1"/>
<smd name="3" x="-2.1" y="-1.6" dx="1.2" dy="1" layer="1"/>
<smd name="5" x="2.1" y="0" dx="1.2" dy="1" layer="1"/>
<smd name="6" x="2.1" y="1.6" dx="1.2" dy="1" layer="1"/>
<smd name="4" x="2.1" y="-1.6" dx="1.2" dy="1" layer="1"/>
<wire x1="-2.5" y1="0.9" x2="-2.5" y2="0.7" width="0.127" layer="21"/>
<wire x1="-2.5" y1="-0.7" x2="-2.5" y2="-0.9" width="0.127" layer="21"/>
<wire x1="2.5" y1="0.9" x2="2.5" y2="0.7" width="0.127" layer="21"/>
<wire x1="-2.5" y1="2.3" x2="-2.5" y2="2.5" width="0.127" layer="21"/>
<wire x1="-2.5" y1="2.5" x2="2.5" y2="2.5" width="0.127" layer="21"/>
<wire x1="2.5" y1="2.5" x2="2.5" y2="2.3" width="0.127" layer="21"/>
<wire x1="2.5" y1="-0.7" x2="2.5" y2="-0.9" width="0.127" layer="21"/>
<wire x1="-2.5" y1="-2.5" x2="2.5" y2="-2.5" width="0.127" layer="21"/>
<wire x1="2.5" y1="-2.3" x2="2.5" y2="-2.5" width="0.127" layer="21"/>
<wire x1="-2.5" y1="-2.3" x2="-2.5" y2="-2.5" width="0.127" layer="21"/>
<wire x1="-2.5" y1="3" x2="-3" y2="2.5" width="0.127" layer="21" curve="90"/>
<text x="-1.5" y="3" size="0.6096" layer="25" font="vector" ratio="20">&gt;Name</text>
<text x="-1.5" y="-3.5" size="0.6096" layer="27" font="vector" ratio="20">&gt;Value</text>
=======
<package name="APA102-2020">
<description>&lt;h3&gt;APA102-2020&lt;/h3&gt;
```

gozhev, 2018-01-25:
I confirm unexpected symbols in LEDS.
There is also dirty merge in Sensors:
`<<<<<<< HEAD
<grid distance="0.05" unitdist="inch" unit="inch" style="lines" multiple="1" display="yes" altdistance="0.025" altunitdist="inch" altunit="inch"/>
=======
<grid distance="0.1" unitdist="inch" unit="inch" style="lines" multiple="1" display="no" altdistance="0.01" altunitdist="inch" altunit="inch"/>
>>>>>>> bdd9b3f71f721249e3a217e2aa19c6b7ac10fbbe
`

squeeler642, 2018-01-26:
Also can confirm I have the nasty merge in Sensors as well. Looking forward to a fix.

designDemon, 2018-04-05:
Hey I faced the same issue and it still hasn't been resolved. A workaround will be to get the old version from GIT. I used a 6 month old version which works!

PaulZC, 2021-07-03:
Hi @uzytkownik ,
It looks like this issue was resolved a long time ago. Certainly the LED opens correctly for me now - in 2021!
A tip is to make sure you are using the actual SparkFun Eagle library from GitHub. We have no control over the Autodesk (adsk) copy of the library and that does sometimes contain bugs.
Best wishes,
Paul

closed 2021-07-03 by PaulZC