sparkfun/SparkFun-Eagle-Libraries #132: SparkFun-LED DRC Error Overlap

renorichard, 2018-02-11:
The SparkFun-LED library parts LED-RGB-CACLEAR, LED-RGB-CADIFFUSE, LED-RGB-CCCLEAR and LED-RGB-CCDIFFUSE all have overlapping pads.

mgrusin, 2018-02-12:
This is a side effect of an obscure setting called "restring." In Eagle 7 and earlier. there's a "restring" tab in the DRC that allows you to shrink pad sizes below what is specified in the library. SFE used this setting to "fix" these parts, and if you're using Eagle 7 or earlier you can do the same thing. However it appears that Eagle 8 (and above?) have removed this setting, so we'll have to fix this in the library. Good catch, thanks for letting us know.

renorichard, 2018-02-12:
Thanks Mike. Yes, I'm using Eagle 8 point something - just installed a few
days ago.

Montiey, 2018-09-03:
Has this been fixed? In the latest version, the pads appear smaller under RMB > Open footprint (Which is LED-RGB-THRU for LED-RGB-CADIFFUSE), but are still in fact still overlapping on the PCB and in the component picker preview.

dragonet80, 2018-09-04:
They are not fixed. I downloaded the Sparkfun-led library a couple of weeks
ago, and had the same problem. I use Eagle 9.1.1.

Missatge de Jason Harriot <notifications@github.com> del dia dl., 3 de set.
2018 a les 6:13:

monodesigns, 2019-03-21:
Same issue using Eagle 9.3.0 for Mac.

ayan4m1, 2022-03-31:
I am running into this issue. The odd thing is that if I open the footprint for LED - RGB PTH, I see

[image]

But in the board view using that part, I see

[image]

lewispg228, 2022-03-31:
Your DRU settings are probably changing this. Check your Annular Ring tab settings in the DRC window. You may have to stray from one of our preset DRU settings files to get those to reduce back down to avoid overlap.

sonic2wb, 2024-02-29:
I just created a new post with the same issue still in the library in 2024.  https://github.com/sparkfun/SparkFun-Eagle-Libraries/issues/359