sparkfun/SparkFun-Eagle-Libraries #137: How to wire the ICSP pads for Mega 2560 Foorprint?

Blacksheep70, 2018-10-05:
I am struggeling for days now getting the ISCP pins from the Sparkfun Arduino Mega Board library wired.
Whatever I am trying, I don't get these pins wired. I think the main problem is that these pins are not included in the schematic but only in the footprint.

For my prototype board I would find it lovely if I can put an ICSP connector on it

Blacksheep70, 2018-10-06:
Same for the 5V pins above pins 22/23 and GND pins below CK/CS

edspark, 2019-03-05:
The ISCP pins are the SPI pins represented on the schematic symbol as the PB3(MISO/PCINT3), PB2(MOSI/PCINT2), and PB1(SCK/PCINT1) pins. If you need a good reference you can always check our  _retired_ Mega Pro on our website. 
https://www.sparkfun.com/products/retired/11007

The pins are wired in the schematic and may help you: 
[image]

closed 2019-03-05 by edspark