sparkfun/SparkFun-Eagle-Libraries #143: arduno uno r3 board - outline and pin locations - pin 8 to SCL off by 10 thou!

Harvolt, 2019-03-29:
hi the relative distance between pin 7 & 8 (ard digital pins), and hence 8 to SCL needs increased from the existing 150 thou to 160 thou. PS, switch to METRIC and disband the other daft masonic practices.

PaulZC, 2021-07-03:
Hi @Harvolt ,
Ah ha! I guess you're talking about the UNO_R3_SHIELD footprint?
Yes indeed, D8 is indeed separated from D7 by 150 thou' (or 3.81 if you're not from the USA). I shall fix that directly!
Thanks for pointing this out - and apologies for the two years it has taken to get back to you!
Paul

Harvolt, 2021-07-03:
Better late than never!

All the best,

Bruce  @ harvolt.

PaulZC, 2021-07-03:
Thanks again Bruce,

Fixed in [this commit](https://github.com/sparkfun/SparkFun-Eagle-Libraries/commit/1f264d3f7ad95c501e3381ca055a55c1974e68c8). Three of the R3 SHIELD footprints had the pins in the wrong place.

Best wishes,
Paul

closed 2021-07-03 by PaulZC