sparkfun/SparkFun-Eagle-Libraries #146: Sensor.lbr redefinition of name 'AS726X' in tag <deviceset> error

edgarlz123, 2019-05-30:
When trying to use the sensor library, Eagle 9.4.1 gives the following error:

> "Loading C:/Users/You/AppData/Roaming/Eagle/lbr/adsk.eagle_library_534/SparkFun-Sensors.lbr ... 
Error: line 13036, column 12: redefinition of name 'AS726X' in tag <deviceset>"

Problem is, there exists two definitions for the AST26X deviceset in the library. 

Following this [suggestion](https://forums.autodesk.com/t5/eagle-forum/part-replace-error-crash/m-p/8384241/highlight/true#M18018) I edited the .lbr file and changed one of the <deviceset> definitions to "AS726X_2"

a-d-v-e-n-t-u-r-o-u-s, 2019-08-01:
The same is happening to me with Eagle 9.4.2.

MateusMolina, 2020-04-28:
Solution provided by edgarlz123 worked well!

Newbie2903, 2021-02-08:
Loading C:/Users/arvin/AppData/Roaming/Eagle/lbr/adsk.eagle_library_534/SparkFun-Sensors.lbr ...

Error:

line 13036, column 12: redefinition of name 'AS726X' in tag <deviceset>

I am getting the same error. where can I get the solution ?

nseidle, 2021-02-08:
There is only AS726X [symbol](https://github.com/sparkfun/SparkFun-Eagle-Libraries/blob/master/SparkFun-Sensors.lbr#L11499) and [device](https://github.com/sparkfun/SparkFun-Eagle-Libraries/blob/master/SparkFun-Sensors.lbr#L14428). This seems to be an error with Autodesk's copy of the SparkFun libraries. We don't have control over those files. I recommend you use the files from this repo and remove the auto-installed library files that come with Eagle via the Library Manager.

closed 2021-02-08 by nseidle