sparkfun/SparkFun-Eagle-Libraries #147: Swapped pin numbers on MCP4725 part.

SFEMark, 2019-06-26:
The pin numbering for the MCP4725 Eagle part is swapped for pins 1 and 3 and 4 and 6, respectively. The pinout for the SOT-23-6 package on page 1 of the [datasheet](https://cdn.sparkfun.com/datasheets/BreakoutBoards/MCP4725_2009.pdf) does not match the [schematic](https://cdn.sparkfun.com/datasheets/BreakoutBoards/MCP4725_Breakout_v14.pdf) linked on the product page for the breakout and the pins are also labeled incorrectly in the Eagle part on the board file.

Just to clarify, our part has these pin definitions:

-  VOUT = Pin 3
- VDD = Pin 1
- A0 = Pin 4 
- SDA = Pin 6.

The datasheet: 
- VOUT = Pin 1
- VDD = Pin 3
- A0 = Pin 6
- SDA = Pin 4

This is not high priority since the labels are correct, but probably should be fixed as it can confuse users of both the breakout and the Eagle part.

PaulZC, 2021-07-03:
Hi Mark,
It looks like this has been fixed:
[image]
I think we can safely close this? True, the MCP4725 schematic could do with being updated though... Maybe we should open an issue over there?
All the best,
Paul

closed 2021-07-03 by PaulZC

SFEMark, 2021-07-06:
Updated schematic and Eagle files ZIP are uploaded to the product page. We're all good here.

PaulZC, 2021-07-06:
Excellent - thanks!