sparkfun/SparkFun-Eagle-Libraries #148: Can't select a component from SparkFun-Sensors.lbr

vredzhepov, 2019-07-20:
I tried to select a component from SparkFun-Sensors.lbr but received an error

Loading /home/vredzhepov/.local/share/Eagle/lbr/adsk.eagle_library_534/SparkFun-Sensors.lbr ...

Error:

line 13036, column 12: redefinition of name 'AS726X' in tag <deviceset>

nabudahab, 2019-07-22:
Was it the MPU-6000? Because I'm having the same issue with that sensor.

vredzhepov, 2019-07-22:
I tried to select AD8232

rawzone, 2020-01-26:
Same error for me (Trying to add in the BRM-2508).

jonny772, 2020-05-12:
Same for me

PaulZC, 2021-07-03:
Hi @vredzhepov / all,
I don't see an issue in the actual SparkFun Sensors library. But it looks like you are trying to use the Autodesk (adsk) version of the library? Yes, sorry, we have no control over that and those libraries do sometimes contain errors. Please use the actual SparkFun libraries from GitHub:
https://learn.sparkfun.com/tutorials/how-to-install-and-setup-eagle?_ga=2.160002557.1113259704.1624989320-559261113.1583443041#using-the-sparkfun-libraries
Best wishes,
Paul

closed 2021-07-03 by PaulZC