sparkfun/SparkFun-Eagle-Libraries #153: ILD213T pinout incorrect

mark1010, 2020-02-22:
Pin pairs 5-6 7-8 are transposed. 

**presently**
Input one: pins 1-2 pair with 5-6 (output two)
Input two: pins 3-4 pair with 7-8 (output one)

**expected**
Input one: pins 1-2 should pair with 7-8
Input two: pins 3-4 should pair with 5-6

_The [BOB-09118 Opto-isolator Breakout](https://www.sparkfun.com/products/9118) uses the same part, from a different library and isn't affected_

PaulZC, 2021-07-03:
Hi @mark1010 ,

Yeah, this is a real problem... The BOB-09118 PCB is routed correctly 1:1 2:2. But the BOB schematic on GitHub now uses the ILD213T from the library - with the swapped pins. If I do an "update all", Eagle see the inconsistency on the pin numbering and the update fails.

OK. The correct way to tackle this is:
- update the pinout in the library (it is wrong and needs fixing)
- leave a note to ourselves that the BOB-09118 schematic will need updating too

The BOB copper won't actually change. But it will be necessary to:
- delete the output wires on the schematic
- update the ILD213T symbol with the corrected one
- connect the wires back up again
- check the PCB as Eagle will have kindly deleted at least part of the tracks while the ILD213T output pins were disconnected

I think I'll seek approval from a higher authority before I tackle this one!
Until later,
Paul

PaulZC, 2021-07-06:
Hi @mark1010 ,
Corrected in these commits:
https://github.com/sparkfun/SparkFun-Eagle-Libraries/commit/d4ca8b02654562ead7d4286d17e2df7556e7d690
https://github.com/sparkfun/Opto_Isolator_Breakout/commit/fcf3b555d61089760e7e3c924f89e31108b939b3
Closing...
Thanks!
Paul

closed 2021-07-06 by PaulZC