sparkfun/SparkFun-Eagle-Libraries #157: Micromod library alignment clarification

dazhbog, 2020-12-01:
Hi Guys,

Trying to overlay the M.2 PCB and the M.2 connector but I cannot fully align the silhouette of the 2 parts. See images below:

[image]

[image]

Is this as expected?

dazhbog, 2020-12-01:
One more pic

[image]

PaulZC, 2021-07-03:
Hi @dazhbog ,
@nseidle made some changes to the M.2 tDocu in [this commit](https://github.com/sparkfun/SparkFun-Eagle-Libraries/commit/5213feb8115199457e1d0cdb821bbbe114693f8a).
Can you please download the latest SparkFun-MicroMod.lbr and let us know if this corrects the issue for you?
Many thanks,
Paul

PaulZC, 2021-07-03:
I'm going to close this as I think this has indeed been corrected, but please re-open if there is still an issue.
Many thanks,
Paul

closed 2021-07-03 by PaulZC