sparkfun/SparkFun-Eagle-Libraries #159: Create export or howto for KiCAD

z3cko, 2021-05-14:
The Sparkfun KiCAD repo is not updated anymore,
but can we please have a recent export or howto for KiCAD?
Eagle is proprietary software and cannot be used by many of us.

Patronics, 2023-07-14:
Especially now that Autodesk has announced that Eagle will be discontinued in a few years, this is more important than ever!

z3cko, 2023-07-16:
I think we might have to one-by-one [import the EAGLE schematics into KiCAD](https://siytek.com/how-to-import-an-eagle-schematic-into-kicad/) and from there export. We can maybe try some [semi-automated way to convert from EAGLE to KiCAD](https://github.com/lachlanA/eagle-to-kicad), but no doubt this will be a lot of work.

PaulZC, 2024-02-08:
Please see:
https://github.com/sparkfun/SparkFun-KiCad-Libraries
https://github.com/sparkfun/SparkFun_KiCad_Panelizer
https://github.com/sparkfun/SparkFun_KiCad_CAMmer

Closing...
Best wishes,
Paul

closed 2024-02-08 by PaulZC