sparkfun/SparkFun-Eagle-Libraries #165: Incorrect footprint for Generic 2012 (0805) package resistor

bhavisheythapar, 2021-07-12:
In the library the dimensions of 0805 (Imperial) resistor is incorrect. The footprint in the library is 3mm x 1.6mm. 

[image]

But as per the standard it should be 2.03 mm x 1.27mm (2012) or 0.08" x 0.05" (0805).

[image]

PaulZC, 2021-07-13:
Hi Bhavishey (@bhavisheythapar ),
I think you are confusing the dimensions of the package with the dimensions of the PCB land pattern?
It varies from manufacturer to manufacturer, but, yes, the 2012 metric = 0805 imperial **package** is 2.0mm x 1.25mm typically +/- 0.1mm.
The manufacturer's **land pattern** is typically 2.6mm x 1.2mm. In our library, we oversize this by an additional 0.4mm to account for: variations in package size, ease and accuracy of pick and place positioning; reliability of soldering, etc..
You will find "tight" package dimensions in the library if you need them - especially on 0402.
Best wishes,
Paul

closed 2021-07-13 by PaulZC

bhavisheythapar, 2021-07-13:
Hi Paul (@PaulZC),

Thank you for the explanation. So when ordering a part from a manufacturer there shouldn't be any issues if I order 0805 resistor  when using this footprint? When I look at the manufacture's datasheet for almost all of them they dimension 0805 resistor to be around (2.0mm x 1.25mm).

Regards,
Bhavishey Thapar

PaulZC, 2021-07-13:
Hi Bhavishey (@bhavisheythapar ),
Yes, you can safely use this footprint with 0805 (imperial) resistors. We have used this footprint 100's of times.
We use the _tKeepout_ layer (layer 39) to show how close the resistors can be placed together - for our pick and place machines.
You will see a "DRC" (Design Rule Check) error if you place the resistors too close together (with the _tKeepout_ lines touching).
Best wishes,
Paul

bhavisheythapar, 2021-07-14:
That's great. Thanks @PaulZC