sparkfun/SparkFun-Eagle-Libraries #213: WS2812B-PTH Connections Appear to be Reversed

slash128v6, 2022-04-16:
I am using this part in a design and all the connections appear to be reversed. For example making a connection to DI in the schematic is connected to DO on the part. same for VDD & VSS when comparing to the hookup guide on Sparkfun.com.

[image]
[image]
[image]

SFE-Brudnerd, 2022-04-16:
Hi @slash128v6,

Would you please provide the git commit hash of your library instance? It might help for diagnosis. In any case, the photo you attached looks electrically correct to me, just that the LED outline is reversed. The ordering is still DIN, VCC, GND, DOUT, which is how it is described in the datasheet. I just checked the latest pull of the library and it appears to be correct with the flat on the other side. 

Thanks.

SFE-Brudnerd, 2022-04-18:
Hi @slash128v6,

I spoke with a few of my colleagues, is it possible you're using the built-in libraries that came with F360/EAGLE? From my understanding those may be out of date. Please use this [tutorial](https://learn.sparkfun.com/tutorials/how-to-install-and-setup-eagle#using-the-sparkfun-libraries) to ensure you're using the most up-to-date libraries. If you're still having issues please let us know.

Thanks.

slash128v6, 2022-04-18:
Hi Alex,

I downloaded and am using the footprint from the SparkFun library from GitHub:

***@***.***

Thank you,

Jason

SFE-Brudnerd, 2022-04-18:
Hi @slash128v6,

Please provide the git commit ID using `git rev-parse HEAD` in a command line/git bash window from the SparkFun-Eagle-Libraries root directory. It appears your local copy is now out of date. After digging through the commit logs, the device you're using (from your photo) was updated, consolidated, and removed 2/13/2021, over a year ago. To update, in the same directory as before please use the command `git pull`. This should pull in the changes you're missing.

Thanks.

slash128v6, 2022-04-18:
Hi @SFE-Brudnerd,

I did not use a git pull for this. I downloaded the ZIP file on 04/14/2022, as per the same as the instruction included in the tutorial link previously. However, upon inspection of that ZIP file contents I noticed they were all dated 06/27/2018. I just downloaded the ZIP again and the files now have today's date. The pin connections appear to be correct now, though the LED symbol within the schematic is flipped: cathode side is marked + and anode side of the symbol is marked -. Thank you for the assistance.

[image]
[image]

sfe-SparkFro, 2022-04-18:
LED symbols have been flipped in 10bebcf4ac0f308943b5452a3162b32265df00e8

slash128v6, 2022-04-18:
@SFE-Brudnerd Looks good, thank you!