sparkfun/SparkFun-Eagle-Libraries #243: OpenLog Library

EricAndrechek, 2022-07-01:
Hi, I am pretty new to Eagle and am learning as I go here, but I am trying to make a project in Eagle involving a [SparkFun OpenLog](https://www.sparkfun.com/products/13712). I downloaded these libraries and added them to my Eagle libraries which worked wonderfully, however, I do not see the OpenLog anywhere. I have tried googling to find it elsewhere but can only find the actual schematics for the OpenLog itself, not its footprint that could just be added into another project. 

I was just wondering if there was a way I could go about doing this and contributing, or if it was something that could be done on SparkFun's end. I am pretty new to Eagle (I have used Altium and KiCad before though) and have been struggling to figure out if I could somehow export the schematics and board of the OpenLog to their own library without much success. 

Let me know if this is something that you think is possible and could be done, or if it already has been and I'm just missing it somewhere. Thanks!

PaulZC, 2022-07-02:
Hi Eric (@EricAndrechek ),

To save you some time, I've put an outline together for you. You will find it in the updated SparkFun-Boards lbr. Here's a quick summary of the steps:

- Fork this library
- Create a new branch for your updates (I used "Add_OpenLog_Board_Footprint")
- Download the OpenLog Eagle design from [here](https://github.com/sparkfun/OpenLog/tree/master/hardware)
- Open the brd (the schematic will open too)
- While you have the brd open, open the SparkFun-Boards lbr from your fork of the GitHub repo
- "Add Device" OPENLOG
- File \ Save All
- Library \ Table of Contents
- "Add Symbol" OPENLOG - or "Duplicate" one of the existing symbols
- Edit the OPENLOG symbol: add the six pins, name them and set their "directions"
- File \ Save All
- Library \ Table of Contents
- "Add Footprint" OPENLOG - or again "Duplicate" one of the existing symbols
- Edit the OPENLOG footprint:
  - Put the pins in the correct positions - matching the OpenLog brd
  - Add the board outline in both tDocu and tPlace
  - Add the name OpenLog in tDocu
- File \ Save All
- Library \ Table of Contents
- Duplicate the OPENLOG footprint as OPENLOG_FLIPPED
- Edit the OPENLOG_FLIPPED footprint: reverse the pins and add "(Flipped)" to tDocu
- File \ Save All
- Library \ Table of Contents
- Edit the OPENLOG Device
  - "Add Part" and add the symbol into the black symbol window
  - Add a new variant, select the OPENLOG package
  - Connect the pins to the pads
  - Add a new variant called _FLIPPED, select the flipped package
  - Connect the pins to the pads
- File \ Save All
- Library \ Table of Contents
- Close the library
- Close the board and schematic
- Submit a GitHub "Pull Request" to merge the changes from your SparkFun-Boards.lbr into the Main library

Enjoy!
Paul

closed 2022-07-02 by PaulZC

EricAndrechek, 2022-07-06:
Fantastic, thank you for your quick reply and helpful walk-through on how to do this myself in the future!