sparkfun/SparkFun-Eagle-Libraries #293: The POWER_JACK_SLOT connector footprint is incorrect.

FirefoxSystem, 2023-02-08:
The three pins are misaligned 100 mils to the left of the drawing on the tPlace layer (21) and those on the tDocu layer (51). 
The two PWR and GND@1 pins should be in the horizontal center of the connector and the GND@2 pin on the right side of the connector, out of the drawing. Effectively as in the POWER_JACK_PTH_BREAD footprint but with long pads.

Note that the POWER_JACK_SLOT footprint, unlike the other footprints, contains the trace on the milling layer (46) corresponding to the three pads to allow the exact execution of the elliptical holes during PCB production.

[image]
[image]

PaulZC, 2023-02-09:
Hi @FirefoxSystem ,

It looks like this was corrected some time ago?

[image]

Please check you are using the latest SparkFun Eagle libraries - direct from GitHub. We have no control over the libraries that come pre-bundled with Eagle.

Best wishes,
Paul

closed 2023-02-09 by PaulZC

FirefoxSystem, 2023-02-09:
Being a managed library isn't it automatically updated like other libraries?
Sorry for the question but I'm relatively new to Eagle.

[image]

PaulZC, 2023-02-09:
These tutorials should help get you up and running:

https://learn.sparkfun.com/tutorials/how-to-install-and-setup-eagle

https://learn.sparkfun.com/tutorials/tags/eagle

All the best,
Paul