sparkfun/SparkFun-Eagle-Libraries #359: Pads for LED-RGB-CA are over lapping

sonic2wb, 2024-02-29:
[image]

sonic2wb, 2024-02-29:
[image]

PaulZC, 2024-02-29:
Hi @sonic2wb ,

I am closing this in favour of #132. Have you checked your design rules (DRU) as mentioned in #132?

Best wishes,
Paul

sonic2wb, 2024-02-29:
I tried to change them but I'm not sure how to do that. If someone could
show me how to fix them. My manufacturer caught the error and I think they
fixed it on their end

sonic2wb, 2024-02-29:
Also this was with the default rules in fusion

PaulZC, 2024-02-29:
If you load the SparkFun-2-Layer-Minimum-JLCPCB design rules, the issue goes away.

Download this file: https://github.com/sparkfun/SparkFun_Eagle_Settings/blob/main/dru/SparkFun-2-Layer-Minimum-JLCPCB.dru

Then, in the Board editor, open Edit \ Design Rules. Click Load, select the .dru file, then click Apply:

[image]

I hope this helps,
Paul

sonic2wb, 2024-02-29:
Thanks I'll try that